caedm/wall-ink #8: Username/password/certificate EAP-TLS support for WiFi

tomuk5, 2018-11-22:
Please add support to upload a certificate or define a username/password to attach to wireless with WPA enterprise.
Allows use in restrictive corporate wireless environments and removes common shares secret for all displays

AndersenJ, 2018-11-26:
Do you need username/password-based WPA Enterprise, or do you need certificate-based WPA Enterprise? Those are two different things, and without knowing your usecase I can't say whether this is possible or not.

From what I understand, there is currently a bug in the binaries provided by Espressif that makes username/password-based WPA Enterprise authentication impossible on the ESP8266. It may be possible to implement certificate-based WPA Enterprise authentication though.

AndersenJ, 2018-12-10:
That's too long without responding; I'm closing this issue. Feel free to open it again later.

closed 2018-12-10 by AndersenJ

tzeeth, 2019-09-22:
Hi! I need to try this with Eduroam too. We want to test the device in a classroom of the University and found this; https://github.com/martinius96/ESP32-eduroam
What do you think? Will it work?
Regards!

AndersenJ, 2019-09-23:
I wish I could say yes, but unfortunately that library is designed for the ESP32, not the ESP8266. It is probably impossible to get it working with Wall-Ink on the ESP8266 due to memory limitations, but it's possible to port Wall-Ink to ESP32. This would require a new PCB to be designed and tested, and it would also likely require some code to be rewritten. We don't currently have the resources to do that. Additionally, when we were doing initial testing, we discovered some problems with the ESP32 platform. It uses about twice as much power (not a deal-breaker, since power usage is still quite low). It's a huge pain to solder (still not a deal-breaker, but annoying). Sometimes, when the ESP32 reboots, [it goes into a bad state and quickly drains the battery](https://github.com/espressif/arduino-esp32/issues/796). We decided this was unacceptable, so we went with the ESP8266.

We use eduroam too, but our devices connect to a guest network. Is it possible that you could do that instead?

tzeeth, 2019-09-23:
Hi @AndersenJ 
Really thanks for your fast response. I will ask if they can use a guest network, if they will does any change need to be made? 
Regards!

AndersenJ, 2019-09-23:
@tzeeth Sorry, I spoke in error before. The reason we can't connect to eduroam is because of problems with the firmware's implementation of WPA2-Enterprise. The community is [painfully aware](https://github.com/esp8266/Arduino/issues?utf8=%E2%9C%93&q=WPA2+Enterprise) of this issue. It has nothing to do with memory; I got mixed up because memory is what's holding us back from being able to do TLS.

When you say, "Does any change need to be made", what exactly do you mean by that? If you're asking about signing in to the guest network, I don't know. We don't need to here, but that may depend on your school's implementation of the network. As for the sketch, they shouldn't need to change anything apart from entering in the correct SSID and password.