caedm/wall-ink #9: No wakeup after the deep sleep

koos, 2018-12-27:
After a successful setup, I'm realizing that the screen/chip doesn't wake up from deep sleep after the choosen time (in my case 30 min.). Only pressing the 'reset' button forces to get the new image. I'm running the development hardware for now.

jdeananderson, 2018-12-27:
Koos,  after looking over the circuit for the development board prototype on our wiki, I just realized that I left off a very important part of the circuit, the part that makes it wake up on its own!  Thank you for finding this error.  I will update the documentation.  For now, I will tell you that you will need to place a 470 Ohm resistor between GPIO16 and RESET, which on the diagram on our "Development board prototype" page would correspond to  a 470 Ohm resistor between D0 and RST, the top left pin and the pin on the bottom, 3rd from the right.  Thanks again for the question and I believe this will resolve your problem.  Thanks,  --Dean

jdeananderson, 2018-12-27:
Koos,

I was easily able to replicate your problem here, and added the 470 ohm resistor between D0 and RST, which completely solved the problem.  I have fixed the documentation.  Please let me know if this solves your issue and I will close the issue.  Thanks,  --Dean

koos, 2018-12-28:
It worked perfectly. Thanks a lot!

closed 2018-12-28 by koos