caedm/wall-ink #18: Error Code is sometimes 4294967295

AndersenJ, 2019-01-25:
Sometimes, the error code is listed as 4294967295. I have no idea why.

This happened once when the base url was mis-entered as http://wallink/.groups.et.byu.net/get_image.php.

AndersenJ, 2019-01-25:
I found roughly 20 instances of this happening in our logs from roughly a dozen different devices. There doesn't seem to be much explanation as to why this happens, so the only way I can imagine I'll learn anything is by hooking the device up via serial and logging the filtered output into a file. It may be that some part of the URL other than the base URL is being malformed, but it isn't immediately obvious how that might be happening.

AndersenJ, 2019-02-13:
I've found that this happens much, much more frequently on some devices than on others. Also, I got the following debug information over serial:

```
http://wallink.groups.et.byu.net/get_image.php?mac_address=68C63A874003&firmware=4.02a_debug&error=4294967295&width=640&height=384&voltage=3.24
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
```

There was some other stuff, but this seemed to be most relevant.

Pillmayr, 2019-04-11:
Is it possible that the problem occurs when the web resource takes too long to create the archive?
My demo installation with static content (Hard coded ICAL File) works perfectly on the devices.
When I dynamically load the iCal externally, I get the same error from time to time.

AndersenJ, 2019-04-11:
I think it's possible that that could trigger the error, but I don't think it's the problem in this case. We have roughly 50 of these things running right now, and one of them gets this error more often than all the rest combined - about 2 dozen times per day. I switched it to the static images plugin to test this though; I'll get back to you tomorrow, if I remember.

Pillmayr, 2019-06-27:
For you... i've implemented an own version of the "image" server (php with laravel -> inspired by https://github.com/bajensen/wall-ink-php) so that the ical data for the calendars are prefiltered & cached by a cronjob. during my work on the server, i found out, that the ical files are quite big, as the ical file contains all the calendar data, even the data in the past. In my case, I had a high-traffic meeting room, where many meetings have already taken place. The ICal File has about 1,8mb (!!). I download the file in advance and save the filtered version of the data so that I only have to handle the data of "today", when the wall-inks connects to download the image-archive.
This almost eliminates the problem.