caedm/wall-ink #20: New Waveshare e-paper HATs will not warm boot.

jdeananderson, 2019-02-21:
For some reason, the new Waveshare e-paper hats will not let a wall-ink device boot unless power is removed completely first.  This means that it will not wake-up from deep sleep on it's own, and it won't go into Admin mode without removing the power and restoring power.  Hitting the reset button will cause the light to blink, but it won't go into admin mode.  Removing the Waveshare hat allows the wall-ink device to check in with the web server without any issues.  Replacing a new hat with an older model hat also fixes the issue.  The newer hats have a lever that runs the length of the connector on the back to clamp the small ribbon cable.  The older hats have a plastic slider with tabs that sits under the ribbon cable.  I will be looking into this issue this week.

jdeananderson, 2019-02-21:
This has been isolated to any esp8266, including a development board, when hooked to a new Waveshare Hat AND the reset pin is hooked to GPIO16 through a 470 ohm resistor.  Everything works fine with the old Hats,  and everything works fine with the new hat if GPIO16 is not pulled down to ground through a 470 Ohm resistor.

jdeananderson, 2019-02-21:
As referenced in this discussion:  https://github.com/universam1/iSpindel/issues/59, GPIO16 to RST can be tricky to keep reliable.  the 470 Ohm resistor is very reliable with the old hat with the TXS0108E chip, but not with the new hat that uses the TXB0108 chip.  I have started preliminary testing with a diode and it appears promising.

jdeananderson, 2019-02-21:
After a bit of study and consulting a couple of EE's, I have ordered a package of Shottky diodes.  We decided that the only way to make a resistor reliable was with an external pull-up circuit with how the ESP8266 works.  We have investigated how a Shottky diode would work in the circuit, and we believe it will work.  I'm going to replace the 470 Ohm resistor in 5 units and run them for a week and see if they miss any check-ins.

jdeananderson, 2019-03-12:
This is still an open issue.  I have not been able to focus on this problem, but have renewed efforts today.  It appears that the shottky diodes do not magically solve this problem.  I will investigate further as time allows.  The old HATs seem to still be just fine in our production systems, but the new ones will still not wake from deep sleep.  I have a dev board working with a shottky diode, but the wall-ink pcbs will not come out of deep sleep with a shottky diode.  I'm hitting the breadboard again to figure out what is going on.

jdeananderson, 2019-03-19:
It appears that this has everything to do with which GPIO pins are used, and most likely GPIO0 and GPIO2.  The other strange thing that happens now is that the LED on the ESP8266 stays on for great lengths of time draining the batteries.  One combination that is reproducible is if RST is hooked to GPIO2, when in deep sleep the LED comes on.  When you unhook RST, the light goes off.  Hook GPIO2 back to RST and it comes back on.  

Here are some GPIO mappings that have been tried with varying success.  

WSS = WaveShare Suggested (from their documentation)
WSD = WaveShare Device, their embedded esp8266 device on a epaper hat
WI0.9 = Wall-ink PCB version 0.9
try1 = a new mapping that seems to boot regularly.

WSS seems to boot every other time.  WI0.9 boots only on cold boot.  Try1 seems to boot every time, but still has an issue with the LED staying on for extraordinarily long times during deep sleep.  The investigation will continue as time allows.

Pin order: 3v3 GND DIN CLK CS DC RST BUSY
WSS: 3v3 GND 13 14 15 0 2 4
WSD: 3v3 GND 13 14 15 4 5 16
WIv0.9: 3v3 GND 13 14 5 0 2 12
try1: 3v3 GND 13 14 15 5 2 4

Notes:  

CS was changed to GPIO5 on WIv0.9 because the device would not boot reliably out of deep sleep with pin 15 used for CS.  Changing it to the "non-standard" GPIO5 seemed to clear up all of the trouble.  Running 50 of these things constantly for months has proved this to be very stable with the old WaveShare Hats.

The WaveShare device uses GPIO16 for busy.  On the Wall-ink device, we are using GPIO16 to bring the device out of deep sleep an investigation would have to be done to know if GPIO16 could be re-used reliably.

jdeananderson, 2019-03-29:
I have verified that the issue is with the new Waveshare hats, and can no longer recommend using the Waveshare hats.    I have purchased a couple of hats, or driver boards from GooDisplay, and they work perfectly.  This is going to require a re-design of the hardware which will be forthcoming.  

This is the driver board:
http://www.buy-lcd.com/index.php?route=product/product&product_id=58595

This "hat" or driver board does not contain the voltage buffering circuit on the Waveshare displays that ensure that 5v signals from an Arduino do not damage the screen.  Since the esp8266 uses 3.3v signalling, we don't need that additional chip-- which seems to be causing issues anyway.  Going forward we are going to base our designs around the less expensive GooDisplay DESPI-C02, even though it has less convenient wire terminals.  Updates will be coming to the 3D printed cases and so forth.  The new cases will need to be compatible with the old backs such that new fronts will fit seamlessly with old backs that are already installed on the walls of buildings.

AndersenJ, 2019-04-15:
New cases are pushed to master, but I haven't finished tweaking the adapter to use the new HATs with the old cases yet.

AndersenJ, 2019-04-19:
The adapter to use the new HATs with the old cases has been pushed to the master branch. Now we just need to finish documenting our changes and we can close this ticket.

Pillmayr, 2019-06-14:
Just to be sure: Can I use the GooDisplay DESPI-C02 Hat with the Waveshare E-Ink 7,5 Inch Displays?
Thx

jdeananderson, 2019-06-14:
Yes, the DESPI-C02 works great with the 7.5" black and white displays!  I have built it on a breadboard and it works perfectly. I am waiting for some new boards to come from China so I can build a real device, take pictures and update the documentation.  Since the DESPI-C02's use a different cable interface, I updated the board to use a different cable since I couldn't find a cheap cable that would work.  I should have an update in a week or so.  Thanks!

Pillmayr, 2019-06-14:
Thank you...
I plan to use the cable that comes with the Waveshare Kit (individual wires) anyway. It may be a bit chunky, but it should (hopefully) work.

jdeananderson, 2019-06-14:
I have done the exact same thing, but I have not been able to find that same cable for sale as a solution to purchase outside of the kit.  When you buy just the screen, you don't get that cable, so I wanted to come up with a cheap solution of all easily found off-the shelf parts.  You should be good to go!  If you get something working, send me a picture and I will put a "make" of it on our site if you'd like.  Thanks!

Pillmayr, 2019-06-25:
Hello,
I got the hats today, and the first tests are going well.
but I do not know how the hat adapter should be mounted. either I printed it wrong or it is too small.
I've now screwed the adapter upside down, so it fixes the board and the board no longer slips.
[image]

jdeananderson, 2019-06-25:
When we got the new hats we had to figure out a way to mount them.  I'm still waiting of a shipment of boards from China to document the final revision of the re-make, but this is more-or-less how it will work.  We had to make a new front with the batteries rotated to make everything fit.
[image]
The plastic "butterfly" piece was meant for what you are doing there, but the other way up with the bevels pointed down to hold the hat in place like this:

[image]

I realize there isn't room for your cables, but this was only meant for if we had a unit already built that the existing hat died and the unit was already assembled and needed to be retro-fitted.  Our plan if this unfortunate situation arose was to unsolder the pin headers off the hat and solder the cable directly in so it would fit.  Not ideal, but I was hoping it would never have to do one.  That's why Jonathan made the new layout as seen on GitHub.  If you re-download the parts, you won't need the hat adapter, and there will be room for your cables. 

I hope this makes sense, and you can get it working.  Sorry for the confusion.  I hope to have all the parts in the next week or so and update all the documentation so it is much more straight forward.

junleus, 2019-08-17:
question: 
what pin order if use DESPI-C02

Pin order: 3v3 GND DIN CLK CS DC RST BUSY
WSS: 3v3 GND 13 14 15 0 2 4
WSD: 3v3 GND 13 14 15 4 5 16
WIv0.9: 3v3 GND 13 14 5 0 2 12
try1: 3v3 GND 13 14 15 5 2 4

matyasforian, 2019-09-27:
Sorry it's not clear to me, which is the recommended display & hat now?
GooDisplay DESPI-C02 Hat with Waveshare E-Ink 7,5 Inch Display? Are other parts of the part list up to date with this spec?
Thanks

jdeananderson, 2019-09-27:
Yes, the GooDisplay DESPI-C02 Hat with the Waveshare E-ink 7.5 inch display works perfectly.  The new Waveshare HATs do not work reliably.  As far as the rest of the parts go, everything else is still current, but the problem is a good way of connecting the wiring between the DESPI-C02 and the board.  I have a solution and have bought the parts to work it up and document it, but I have been buried in projects.  It just needs to get done and I will do it either this afternoon or Monday.  I will update the parts list when I get the new pictures up in the next few days.  Thanks for asking, and thanks for your patience.

AndersenJ, 2019-10-17:
In case anyone was wondering where this issue is, we discovered we don't currently have the parts to build the new board. We have the PCB, just missing some miscellaneous parts. We'll get this updated once we've ordered the parts and actually put it together.

AlexandreBlais42, 2024-05-01:
I was able to adapt the circuit to make it work with the new HAT (Rev 2.2) by changing R1 to 100 Ohms and R8 to 470 Ohms. It was able to warm boot for the whole night.