caedm/wall-ink #21: wall-ink-server running https

zhongshanliyu, 2019-08-19:
when wall-ink-server running https, the wall-ink get error message when get https://example.com/get_image.php.

zhongshanliyu, 2019-08-19:
there are a example to showing the HTTPSRequest, i thiank that can solve the issue.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino

zhongshanliyu, 2019-08-19:
error code wall-ink-server logs below 

Firmware: 4.02a Width: 640 Height: 384 Voltage: 2.85 Error Code: 4294967295 Plugin: 1

Pillmayr, 2019-08-19:
Hello. As you said before briefly: The firmware does not use the Secure Client but the normal Wifi client. I would simply customize the firmware in your place and use the Secure Client instead of the normal client: https://github.com/caedm/wall-ink/blob/master/stream_screen/stream_screen.ino#L730
The second option would be to instantiate the client based on the protocol. But I think that the first variant is more pragmatic and faster to implement, because all of your endpoints are running on HTTPS anyway, right?

jdeananderson, 2019-08-19:
We investigated using an https client, but if I remember correctly there was not enough free memory in the sketch on the esp8266 to support it.  Perhaps Johnathan remembers?

zhongshanliyu, 2019-08-20:
> Hello. As you said before briefly: The firmware does not use the Secure Client but the normal Wifi client. I would simply customize the firmware in your place and use the Secure Client instead of [...]

correct, wall-ink-server running internet, and wall-ink get https url will encrypt the mac address etc.

zhongshanliyu, 2019-09-07:
[image]
that true, memory not enough to running.

closed 2019-09-07 by zhongshanliyu