caedm/wall-ink #25: Find alternative for e-ink screens

AndersenJ, 2019-11-26:
As described in https://github.com/caedm/wall-ink/issues/24, the screens we're currently using aren't likely going to be easy or even possible to purchase in the future. We'll need to find a replacement screen. We may need to change some code (or in the worst case, the case) to accommodate that new screen choice.

ljorzick, 2020-07-26:
Is the v.2 of the originally used waveshare display an alternative? https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT
Should be compatible in terms of hardware, but will require a code change due to changed resolution...or are we hitting memory boundaries on the ESP with the increased resolution (seen this mentioned in the project somewhere).
Background: I'm considering this as a student project for a course at my school, but am currently looking to source parts for a proof of concept and couldn't source the original display...

AndersenJ, 2020-07-26:
It's hitting memory boundaries. The ESP8266 doesn't have enough RAM for this. That said, I think this screen does support partial updates. This means you could modify the firmware and server software so it pulls down only half of the image at a time. This would probably not be too difficult.

ljorzick, 2020-08-02:
Hmm,
I'm more of a networking guy with not much in-depth knowledge in terms of coding. Are you investigating any replacement options which will work with the project ootb or is there any perspective in that regard? Thanks for your time and effort, the project looks amazing :-)

AndersenJ, 2020-08-03:
I actually don't work for CAEDM anymore so I can't speak for them, but I'm fairly sure that finding a replacement option is a very low priority for them right now. It may actually never happen. I may decide to give it a try on my own time in a few months though, just for fun.

DIYglenn, 2023-10-04:
How much work would it be to make it work with ESP32? It should solve both this issue, and could give both better security functionality and possibly color options. Color could work as long as they support partial updates, otherwise some are really slow.

AndersenJ, 2023-10-05:
It's quite doable to make it work with the ESP32, and it would indeed solve the issue, but it also introduces a new issue: the ESP32 doesn't consistently wake from sleep. Sometimes instead of waking properly, it'll just go into a state where it drains the batteries while doing nothing. Here's some detail: https://github.com/espressif/arduino-esp32/issues/796

That said, it's been a while since anyone worked on this. It's certainly possible that there's some other new product that would work better.

DIYglenn, 2023-10-06:
That's true. But it does open for a lot of possibilities for devices connected to power, (mains or PoE f.ex). 
Then you'd have partial update possibilities etc. So while it does consume more electricity, and is not as flexible to install as before, you'd still have an extremely low draw for an always on screen.