caedm/wall-ink #30: FAQ: Can I buy this as a full product on Kickstarter?

fulldecent, 2020-08-10:
Surely this is a top question people will want to know.

jdeananderson, 2020-08-10:
We did expect this to be a frequently asked question.  We included the answer In the readme.md, which is the homepage for the project.  I have copied the answer here for your convenience:

Can I hire you to make the hardware for me?

Due to conflict of interest policies at the University we work for, we are not available for hire to create hardware for you, or to customize software features for you. That being said, we have uploaded all of the design files and code, and reasonable documentation to do it on your own. Many people have created this project on their own with positive results.

fulldecent, 2020-08-11:
Super helpful, thank you. I will look out on eBay and wait for some enterprising person to get them up there.

closed 2020-08-11 by fulldecent